RAKWireless/Products_practice_based_on_RUI #1: API_Reference download link is not working

salmanfarisvp, 2019-09-06:
The link for the **API_Reference** is not working  [https://downloads.rakwireless.com/en/RUI-API/RUI_API_Reference_V1.0.pdf](https://downloads.rakwireless.com/en/RUI-API/RUI_API_Reference_V1.0.pdf)

and Is there any other resources for RAK5010 module?

RAKWireless, 2019-09-06:
Hi Salman,

RUI APIs has been placed here:
https://doc.rakwireless.com/developer-tools/-/-

For RAK5010, we'll update a new firmware and document next Tuesday.

Best Regard
________

Fomi Tong
E-mail: fomi@rakwireless.com

深圳市瑞科慧联科技有限公司 Shenzhen RAKwireless Technology Co.,Ltd.
深圳市南山区桃源街道平山一路新视艺创客公园B栋506室
Room 506, Bldg B, New Compark, Pingshan First Road, Taoyuan Street,XiLi town Nanshan District, Shenzhen, China
Website: www.rakwireless.com  |  微信公众号/官方微博：RAK瑞科慧联
Twitter: @RAKwireless  |  YouTube：RAK YouTube Channel
 
From: Salman Faris
Date: 2019-09-06 16:51
To: RAKWireless/Products_practice_based_on_RUI
CC: Subscribed
Subject: [RAKWireless/Products_practice_based_on_RUI] API_Reference download link is not working (#1)
The link for the API_Reference is not working https://downloads.rakwireless.com/en/RUI-API/RUI_API_Reference_V1.0.pdf
and Is there any other resources for RAK5010 module?

salmanfarisvp, 2019-09-06:
Thanks , that was very quick!

closed 2019-09-06 by salmanfarisvp