RAKWireless/Products_practice_based_on_RUI #2: how about setting interrupt for gpio?

letruongthanh24103698, 2019-10-29:


RAKWireless, 2019-10-29:
Actually, we're working on it, and RAK5010 has supported this new API now, but other modules haven't supported it. So if your customization is based on RAK5010, you can use this API now, but if not, maybe you need to wait 1~2 weeks for our update.

/***************************************************************************************
 * @brief       This API is used to regist external gpio, it may lead to power current 
                increase. e.g 130uA in nordic platform.
 * @return      RUI_RETURN_STATUS
 * @param       bool control: true or false, use it or not.
                RUI_GPIO_ST st: gpio struct
                edge: detect in raise or fall
                pro: priority for interrupt
                callback: interrupt callback.
 * @module		RAK811	 RAK4200 	RAK4600 	RAK4400  RAK5010 
RAK8212(M)   
***************************************************************************************/
typedef void (*interrupt_callback)(void);
RUI_RETURN_STATUS rui_gpio_interrupt(bool control, RUI_GPIO_ST st, RUI_GPIO_INTERRUPT_EDGE edge, RUI_GPIO_INTERRUPT_PRIORITY pro,interrupt_callback callback);

letruongthanh24103698, 2019-10-29:
unfortanately my customization is based on RAK811. Hope to see the API soon

RAKWireless, 2019-10-29:
OK, we'll update it ASAP.

closed 2019-10-29 by letruongthanh24103698

Navid007, 2020-03-26:
Hello
Does anyone know how to disable interrupt ? 
placing "false" in below command does not disable interrupt. Is there any other command?

rui_gpio_interrupt(false, Interrupt_IO, RUI_GPIO_EDGE_RAISE, RUI_GPIO_IRQ_HIGH_PRIORITY, interrupt_routine);