RAKWireless/Products_practice_based_on_RUI #4: ..//Source/stm/STML151/board/board.c:16:10: fatal error: rui.h: No such file or directory

x893, 2020-02-16:
Hi,
i try compile example application for RAK811-H from app_RAK811 folder.
Compiler log:
mkdir _build		
	CC	adc-board.c
	CC	board.c
	CC	sx1276-board.c
	CC	timer-board.c
	CC	rtc-board.c
	CC	i2c-board.c
	CC	gpio-board.c
	CC	spi-board.c
..//Source/stm/STML151/board/board.c:16:10: fatal error: rui.h: No such file or directory
 #include "rui.h"          ^~~~~~~
compilation terminated.
stm/Makefile_stm32l151cb:246: recipe for target '_build/board.o' failed
make: *** [_build/board.o] Error 1
make: *** Waiting for unfinished jobs....

Best wishes
x893

RAKWireless, 2020-02-17:
Hi,
I think you may forget to copy the rui.h file into your project folder. The rui.h file can be found in the folder <common header> of our github repo.

x893, 2020-02-17:
Yes, after add rui.h all completed.
Thanks

closed 2020-02-17 by x893