RAKWireless/Products_practice_based_on_RUI #5: RAK4200 ADC

letruongthanh24103698, 2020-04-24:
Hi, i'm using rak4200 module im my own dev board. I config PA3 as ADC INPUT because according to this https://www.st.com/resource/en/datasheet/stm32l071v8.pdf, PA3 has ADC_IN3. but in https://downloads.rakwireless.com/LoRa/RAK4200/Hardware-Specification/RAK4200_Module_Specifications_V1.4.pdf PA3 does not have ADC. Furthermore, when i try this command at+get_config=device:adc:X with x 1->9, only 3 working.
So, which pin i can use for adc?
Does the RAK4200_Module_Specifications_V1.4 right?

RAKWireless, 2020-04-26:
Only pin 3 supports ADC in rak4200. Pin 3 is PA1 in stm32l071, and we use ADC_IN1 . So in the command at+get_config=device:adc:X, parameter 3 works.

letruongthanh24103698, 2020-04-27:
So, "Rak4200 RUI only support ADC in PA1" or "RAK4200 only support ADC in PA1"