RAKWireless/Products_practice_based_on_RUI #7: How to enable UART with DE in RUI SDK

KunYi, 2020-05-03:
Hello, 
    I have new questions for RAK4200,

1. Would you tell me how to configuration UART2 with RS485 Driver Enable?
2. What's difference for [UART mode](https://doc.rakwireless.com/developer-tools/developer-tools/rui-uart-configure-mode)?
- RUI_UART_NORMAL
- RUI_UART_UNVARNISHED

RAKWireless, 2020-05-03:
Hi @KunYi ,

RUI_UART_NORMAL is the AT configure mode which means you can send AT command to RAK4200 now, and RAK4200 will deal with every valid AT command.
RUI_UART_UNVARNISHED is the data transmission mode which means you can send any data into RAK4200 through this UART and RAK4200 will receive it. In this mode, even if you send an AT command, RAK4200 will deal with it as a data.

About the question of UART2 with RS485, there isn't RS485 driver integrated now, if you want, you can add this feature by yourself by usng RUI APIs about UART:
https://doc.rakwireless.com/developer-tools/developer-tools/rui-uart-send
https://doc.rakwireless.com/developer-tools/developer-tools/rui-uart-receive

KunYi, 2020-05-03:
@RAKWireless , 
thank your for you reply

but about question 2.
in STM32L0 HAL have HAL_RS485Ex_Init function for RS485-DE
How to apply the functions in RUI SDK?

KunYi, 2020-05-03:
okay, maybe some information missed,

please check your RAK4200 user manual about UART1-DE and UART2-DE (DE: Driver Enable)  and check your RUI SDK/STM32L0Cube about HAL_RS485Ex_Init()
I suggestion add a hook function/(weak attribute) in usart_board.c
for user customization